bambulab/BambuStudio #5551: top shell layers

robinehoward, 2024-12-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 home 10.0.19045

### Additional system information

intel i5-10400
memory 8gb 
intel UHD Graphics 630

### Printer

bambu labs P1S/A1/A1 mini

### How to reproduce

slice a file

### Actual results

top layers set to 2. I slice and end up with 1 bridge/5 solid infill and 1 top layer. change to 1 top layers set to 2. I slice and end up with 1 bridge/5 solid infill and 1 top layer. change to 3 top layers set to 2. I slice and end up with 1 bridge/5 solid infill and 1 top layer

### Expected results

set in the number of top layers and seeing a change in the number of solid infill layers

### Project file & Debug log uploads

[solarLightMountPl.zip](https://github.com/user-attachments/files/18141495/solarLightMountPl.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2025-04-02:
[image] 
this setting will increase the layer number to ensure it had enough thickness. set this to 0, than you will get what you want

QingZhangBambu, 2025-05-23:
close due to no response

closed 2025-05-23 by QingZhangBambu