bambulab/BambuStudio #5557: Trying to export Gcode with Creality K1C

Draketornado, 2024-12-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

My dad is owning a Creality K1C and he tryed to export a Gcode. Will not work for thee K1C, but it works for other printers.

Error:

Verarbeitung der Vorlage filename_format fehlgeschlagen.
Parsing error at line 1: Not a variable name
{input_filename_base}_{filament_type[initial_tool]}_{print_time}.gcode

### Actual results

Error: Verarbeitung der Vorlage filename_format fehlgeschlagen.
Parsing error at line 1: Not a variable name
{input_filename_base}_{filament_type[initial_tool]}_{print_time}.gcode

### Expected results

Should open a window to safe the gcode

### Project file & Debug log uploads

Tryed with a lot of files, never worked

### Checklist of files to include

- [ ] Log file
- [ ] Project file

00raq00, 2024-12-18:
Try change to 

{input_filename_base}{filament_type[0]}{print_time}.gcode

Draketornado, 2024-12-22:
Thx for the reply.

For me, i cannot input this in BambuStudio as this data field is not present, but i can change this value in OrcaSlicer. The weird thing is that in OrcaSlicer, the value is {input_filename_base}{filament_type[initial_tool]}{print_time}.gcode and it's working!

00raq00, 2024-12-22:
> Thx for the reply.
> 
> For me, i cannot input this in BambuStudio as this data field is not present, but i can change this value in OrcaSlicer. The weird thing is that in OrcaSlicer, the value is {input_filename_base}{filament_type[initial_tool]}{print_time}.gcode and it's working!

Try to turn on advanced view in Bambu

Draketornado, 2024-12-23:
Advanced view is activated, but this line does not show up like in Orca Slicer. I even tried to switch language, without result.

XunZhangBambu, 2024-12-29:
@Draketornado   It's on the develop mode . You can go to "preference" page to enable develop mode and then you will see the lines

closed 2025-06-03 by XunZhangBambu