bambulab/BambuStudio #5562: Bambu Studio issue with text shape option

Sere994, 2024-12-16:
### Bambu Studio Version

v01.10.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

BambuLab A1

### How to reproduce

1. open a project
2. select the text option
3. here before the update i've been able to check both of the options in the bottom (surface and orizontal text) and now i can only chose one.

### Actual results

here before the update i've been able to check both of the options in the bottom (surface and orizontal text) and now i can only chose one. 
I made a gif for explain the situation before. Before this update i was able to make the text rounded checking both the options, now if i try to do the same i can't check both options so the text remains flat o not in the surface. 

[image]

### Expected results

If i try to open this file now and modify something in the text it became flat.

### Project file & Debug log uploads

[Flying Disk Fidget KeyChain_ AMS _ Only Text.zip](https://github.com/user-attachments/files/18153276/Flying.Disk.Fidget.KeyChain_.AMS._.Only.Text.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-12-18:
Hello, we have fixed the issue. Please wait for the next version. At present, you can temporarily replace the editor with an old version~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00