bambulab/BambuStudio #5563: Mesh Boolean causing Bambu Studio crash

mattboonecochran, 2024-12-16:
### Bambu Studio Version

7.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu P1S w/AMS

### How to reproduce

Just try to use Mesh Boolean for parts in an assembly.

### Actual results

Whenever i use Mesh Boolean, its about an 80% chance that it will crash the program. Waiting for it to respond is less than a 50/50 shot, i usually have to restart the program and try again, hoping that i remembered to save my progress recently. I have resorted to just not meshing parts most the time, In the current project, it has taken me 30 minutes to mesh 8 parts.

### Expected results

parts should mesh within a respectable time (maybe a second or two) without the program crashing........

### Project file & Debug log uploads

the compressed zip file is 31mb, apparently too big to attach here...its just one emblem about 3"x4". so i guess this is just a bug report for y'all to look into.

### Checklist of files to include

- [ ] Log file
- [X] Project file

mattboonecochran, 2024-12-16:
i thought it automatically populated the additional system info box but i guess it didnt. so ill add it here.

lenovo yoga pro 9
processor: Intel(R) Core(TM) Ultra 9 185H   2.30 GHz
ram: 32GB
Graphics: NVIDIA GeForce RTX 4050 Laptop GPU

Haidiye00, 2024-12-23:
Hello, we will release a version in the future that will detect some errors in advance and try to ensure that the mesh boolean runs without crashing~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00