bambulab/BambuStudio #5570: Bambu Studio Device Status Page Fails to update on color change

tmclink, 2024-12-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.7.1

### Additional system information

2.3 GHz 8-Core Intel Core i9, AMD Radeon Pro 5500M 4 GB

### Printer

X1 Carbon

### How to reproduce

Initiate a multi-color print, go to Device Status tab and wait for color change.   The line in the AMS section does not update to the new color when the filament changes (however a small portion of the line near the print head symbol does change)

### Actual results

No color change in depiction of selected color

### Expected results

Color change matching the currently selected color

### Project file & Debug log uploads

[debug_Sun_Dec_15_12_46_19_67995.log.0.zip](https://github.com/user-attachments/files/18180971/debug_Sun_Dec_15_12_46_19_67995.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

tmclink, 2024-12-19:
After a filament change, then screen updates incompletely:  [image]

After leaving the tab and coming back: 
[image]

walterwongbbl, 2025-01-01:
@tmclink 
This seems to be related to the screen refresh rate; it occurs on some computers, and we are working on a solution.