bambulab/BambuStudio #5572: Generated G-Code for a1 mini, not sure about other printers

oleksharh, 2024-12-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab a1 mini

### How to reproduce

1. Download attached file
2. Generate g-code
3. Run Print on the printer itself
4. On the overhangs greater than specified angle of the overhangs, fan speed starts kicking in only mid printing small-mid sized overhangs, causing them to stick to the nozzle when printing with PETG and not sticking well to previous layers.
[Christmas+Tree+for+LED+ a1m+kit(2).zip](https://github.com/user-attachments/files/18184099/Christmas%2BTree%2Bfor%2BLED%2B.a1m%2Bkit.2.zip)

### Actual results

On the overhangs greater than specified angle of the overhangs, fan speed starts kicking in only mid printing small-mid sized overhangs, causing them to stick to the nozzle when printing with PETG and not sticking well to previous layers.

### Expected results

Fan on the printer should start working slightly before entering the overhang to accumulate needed speed to cool down the overhang so it turns out straight and not warped and not well-sticked to the last layer

### Project file & Debug log uploads

[Christmas+Tree+for+LED+ a1m+kit(2).zip](https://github.com/user-attachments/files/18184102/Christmas%2BTree%2Bfor%2BLED%2B.a1m%2Bkit.2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ScorpioLingDong, 2025-02-25:
Hello, thanks for your suggestion! We are already working on it. The fan will pre-start 2 seconds before encountering an overhang line. Based on our tests, it works well and will be included in the next version release.

ScorpioLingDong, 2025-02-25:
Hello, thanks for your suggestion! We've implemented this feature and will be included in the next version release.