bambulab/BambuStudio #5575: Interface artifacts

jvalal, 2024-12-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Apple - Seqoia 15.1.1 (24B91)

### Printer

A1

### How to reproduce

Open the software and go to the devices tab. You can see the numbers have lines through them.

[image]

### Actual results

See above

### Expected results

Expect the text to be legible like the other text. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

techjoh, 2024-12-19:
i also have these artifacts on my bambustudio. -> v1.10.1.50

DanBao-Bambu, 2024-12-25:
Thank you for the report. We will fix the issue in the future.