bambulab/BambuStudio #5576: buggy device stream in bambu studio

techjoh, 2024-12-19:
### Bambu Studio Version

v1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.2

### Additional system information

M4 - Macbook Pro

### Printer

BambuLab A1 Mini

### How to reproduce

open BambuStudio
goto Device
right down corner is a failure text:
player is malfunctioning. please reinstall the system player.
[image]

### Actual results

there is no video stream from actual printprogress available since the firmware update from the a1 mini.
happens on all printprofiles. reopening the app doesnt help. also restarting the whole system. also reinstall bambustudio doesnt solve the problem.

### Expected results

there should be a videostream from an actual printprogress.
iOS App works still perfect. but on MAC there is this failure for me

### Project file & Debug log uploads

[Warehouse Label v1.3.3mf.zip](https://github.com/user-attachments/files/18195217/Warehouse.Label.v1.3.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

techjoh, 2024-12-19:
also the SDCARD ShowTab doesnt work anymore with macOS.
[image]

bambu123, 2024-12-23:
Same As #5010

djblue81, 2025-01-17:
Micheal Jackson says: You are not alone