bambulab/BambuStudio #5577: zoom

sherrahs, 2024-12-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 Mini

### How to reproduce

When I touch the gray area with my mouse and try to work on a plate or add files, it keeps zooming all the way in.  It zooms until the entire screen is whitish gray. I can only get it to stop zooming by moving the mouse to the right, or up to the information tabs, or down at the bottom, or totally off of the bambu app.  It just stared to do this yesterday.  I have been using it for months with the A1 mini and it has never done this.   I have already made sure that the zoom feature in preference tab is off, and I have already installed latest update.  I have also, obviously, closed the program, and attempted different files.  Its still doing it.  I have also cleaned my mouse, my touchscreen, and my touch pad on my keyboard.  Its only zooming as I described in bambu studio.  It is not zooming automatically on any other program or use of computer.  Its just in the studio.  Can you help me?  

### Actual results

Sorry, cant screen shot it happening, it just freezes for its screenshot when I do that.  And shows that the plate is still.

### Expected results

It should have stopped zooming :) 

### Project file & Debug log uploads

there is no file in particular.  it is doing it with every file I load

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-12-23:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
(https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-04-08:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00