bambulab/BambuStudio #5578: mac中文本形状，无法输入中文内容（复制可以），但也很多乱码

huanyan7, 2024-12-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

mac 14.6.1 (23G93)

### Printer

p1p

### How to reproduce

1.文本形状
2.输入文本

### Actual results

[image]

### Expected results

能写中文

### Project file & Debug log uploads

什么鬼东西。

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-12-23:
Hello, we are currently fixing this issue

kindansljd, 2025-01-24:
请问，这个BUG什么时候能被修复呀

Haidiye00, 2025-01-24:
Hello, I'm very sorry, it may be in March, and we are currently unable to determine the release date of the version.

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~