bambulab/BambuStudio #5583: Status Screen does not update properly

Jeffb9, 2024-12-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Processor	Intel(R) Core(TM) i5-3340 CPU @ 3.10GHz, 3101 Mhz, 4 Core(s), 4 Logical Processor(s)
Installed Physical Memory (RAM)	12.0 GB

### Printer

P1S

### How to reproduce

When preparing to print, AMS/Print Head loads, filament trace does not show from AMS to grey block above print head icon. Does show from grey block to print head. If I get out of screen and return, displays properly.
Same action when unloading print head. Switch from STATUS screen to any other and return, display updates properly.

### Actual results

See above, screen does not update.
[Black filament loaded.pdf](https://github.com/user-attachments/files/18203584/Black.filament.loaded.pdf)

### Expected results

Screen should update in realtime.

### Project file & Debug log uploads

Not applicable to this issue.

### Checklist of files to include

- [X] Log file
- [X] Project file

tmclink, 2024-12-19:
I have this issue also

walterwongbbl, 2025-01-07:
Hi.@walterwongbbl @tmclink 
This issue is due to the reduced refresh rate, and we will work on fixing it.