bambulab/BambuStudio #5589: 0.6 crashes on 0.24 and 0.18

Ofr3d, 2024-12-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11, 23H2, 22631.4602, Windows Feature Experience Pack 1000.22700.1055.0

### Additional system information

12th Gen Intel(R) Core(TM) i7-12700K   3.60 GHz,
64.0 GB,
64-bit operating system, x64-based processor

### Printer

P1S

### How to reproduce

0.6 hardened steel, 0.24 and 0.18 layer height.

[image]
ht.

### Actual results

Software crashes every time.

### Expected results

Print

### Project file & Debug log uploads

[Board.zip](https://github.com/user-attachments/files/18212927/Board.zip)
[Board 0.18.zip](https://github.com/user-attachments/files/18212953/Board.0.18.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-12-21:
HI [Ofr3d](https://github.com/Ofr3d)：Do you mean that studio  crash when you click slice button

Ofr3d, 2024-12-21:
Hello,
Exactly what it means,  I load a model, choose 0.24 or 0.18 layer hight (on
a 0.6 hardened steel nozzle) and I press slice, the model slices until line
800 or so and then the software crashes and I have to restart it. I had to
print in 0.3 layer hight because the program did not finish the slicing  in
0.18 nor 0.24.

QingZhangBambu, 2025-04-02:
@Ofr3d could you try the latest version? i've slice the file on latest version successfully. If it not work, could you upload dmp files？i could not reproduce this issue