bambulab/BambuStudio #5590: AMS Auto Refill - Spool Run Out - Wrong Colour

stu583, 2024-12-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS sequoia (v15.0)

### Printer

Bambu Lab P1S

### How to reproduce

Can really reproduce apart from printing. 

I have x3 AMS's interlinked. All working great except Auto Refil problem. It is switched on but is identifying BambuLabs PLA Black and Green as the "same" thing. When black in position A1 (black) runs out it is changing to B3 (green) and ruining the colour in prints.

### Actual results

1. Spool Runs out in Position A1 (BambuLabs PLA Black)
2. AMS loads Filament B3 (BambuLabs PLA Green)
3. Continue Printing with wrong colour### ### 

### Expected results

1. Spool Runs out in Position A1 (BambuLabs PLA Black)
2. AMS Looks for another reel matching Brand: BambuLabs, Type: PLA, Colour: Black
3. AMS unable to find another appropriate filament
4. Pauses, displays run out warning, waits for user input

### Project file & Debug log uploads

Happens with any multicolour model but only when filament runs out. Colour changes are working fine until a spool runs out.

### Checklist of files to include

- [X] Log file
- [X] Project file

gaolegao-lx, 2024-12-23:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

jeglover, 2024-12-31:
For what it's worth, I just ran into this issue as well and though I was going insane. I reset my printer (and AMS units), restarted Bambu Studio, and did anything else I could think of and it's still seeing PLA Basic Black and Green as the same thing -- all three are official Bambu Labs spools. (See attached image) 

[image]

The system is correctly interpreting the spools themselves, it's just that it considers the two blacks and one green to be identical for auto-refilling purposes.

thinkingalaud, 2024-12-31:
My situation is very similar, except I don't use Bambu Lab spools - I only use Polymaker's PolyTerra line. This only happens to me with black and green colors, and both are custom colors. I also have a P1S, Bambu Studio v1.10.0.89, macOS Sequoia v15.1.1.

Really annoying because I was printing in green and relying on auto refill to notify me when I ran out of filament so I could replace the spool, but the printer instead automatically started using black and ruined the print.

jorenvanlindert, 2025-01-08:
Same issue here, it seems specific to black and green somehow.
[image]

volltz, 2025-01-16:
I am in a similar situation and have even two different groups which do not match at all:

[image]

Even worse: The Filament B2 is not even PLA but it is the special "PLA for Support", so if it would ever switch to this one the print would not only be optically ruined, but would also fail

JoaquinVelascoJ, 2025-02-08:
I have the same issue with a P1S with AMS and Bamu Studio. I have one blue (A4) and one black (A2) bambulab PLA Matte, and I'm trying to print only blue objects. The auto refill seems to suggest it will continue printing with black after blue runs out... 
I haven't tested if that would actually happen if blue runs out

scallister, 2025-02-18:
An example showing the `Jade White` Filament properly grouping while the `Black` filament is getting assigned other color types.

[image]

Merovech, 2025-02-21:
I'm running into this now on my A1.  As of 10 minutes ago it suddenly started seeing a new PLA spool I put in to replace an empty one as part of the group, even though they're not the same color.  All three are Bambu spools.

[image]

BrainStone, 2025-03-22:
Also happening for my A1 with AMS lite.
Using Version 1.10.2.76 on Ubuntu 22

[image]

1980alexGS, 2025-03-23:
And me, P1S with AMS. I'm glad there are others with the same problem (black and green considered equal) as it was driving me crazy. When A1 runs out, I want it to switch to A2, and then if I print several more times and A2 runs out, I want it to switch to A1. It should not switch to A3. I seem to have no control over this 'Group1'

[image]

4Flat4Life, 2025-04-15:
Adding to the list... 
My PLA Basic Black ran out (Slot 3), I have another roll of PLA Basic in slot 4 and it switched to slot 2 which is PLA Basic Cyan... 
Ruined the print.

Patralos, 2025-04-16:
Same issue for me

davidounios, 2025-04-16:
[image]

2.0.1.50 Latest version of OSX M4 cpu.

There was matte black in B1 then it swapped to matte marine? blue.

Sector14, 2025-07-01:
Just happened to me, printing matte green which ran out and I was expecting it to then pause and ask me to add more (which I don't have) or select a different colour, would have selected white to finish the print. Instead, it auto switched to the "same colour" and printed in black...

Version 2.0.3.54 on Windows.

davidounios, 2025-07-02:
`    if (!f->color.empty()) {
                    ams_col = AmsTray::decode_color(f->color);
                } else {
                    // default color
                    ams_col = wxColour(0xCE, 0xCE, 0xCE);
                }`

Had a quick squiz at the codebase. Looks like if they aren't getting any color info at this point it's setting it to a default value. Probably a very rare scenario where it can't get the colour at this point when mapping but still does later when showing colour in the UI. Would be much better having it as 'unknown' then checking for that during the auto_refill stuff.