bambulab/BambuStudio #5591: "Flow Calibration" not possible

jensrad, 2024-12-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Start Bambu Studio
2. Go to menu "calibration" => "flow calibration"
3. scroll to bottom
4. there are two buttons: "Manage Results" and "Manuell calibration". 

[image]

### Actual results

Both buttons are grayed out, so I cannot start flow calibration. On the other hand: "Flow Rate" calibration is possible

### Expected results

Expectation is that "Flow Dynamik Calibration" can be started manually or automatically according Bambu Wiki (https://wiki.bambulab.com/en/software/bambu-studio/calibration_pa)

### Project file & Debug log uploads

no log or project file available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-12-21:
Thanks for reporting this issue, we will fix it in the future. 
You can switch machines or reconnect the machine on the device page to avoid this problem.

Bandy1284, 2024-12-28:
I have the exact same issues, this has only happened since the last big update. Not the recent one with just filaments added. 
Its incredibly frustrating as Ive just replaced a nozzle and ideally would run the test quickly before each filament use.

we-rebel-scum, 2025-01-16:
A bit frustrating, but [this workaround does work](https://www.reddit.com/r/BambuLab/comments/1gpqypp/comment/lx0nwon/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button). You just need to be very quick to catch the buttons. The first time I did this, it showed "Incompatible" for the filament, but I could set it from my list. Selecting the radio button allowed the calibration to proceed.

[image]

zhimin-zeng-bambulab, 2025-02-05:
It has been fixed in v1.10.2 Public Beta

closed 2025-02-05 by zhimin-zeng-bambulab