bambulab/BambuStudio #5593: Flickering

mpl1337, 2024-12-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11 

### Additional system information

-Intel(R) Core(TM) Ultra 5 245K with GPU
-32GB DDR5 RAM

### Printer

P1S

### How to reproduce

Use the Bambustudio with and without Modell

### Actual results

The Window begins to Flickerin until i stop it. Sometimes its stop and beginn again after some clicks in Bambustudio.

### Expected results

This flickering only happens with bambustudio. Other Software Works fine.

### Project file & Debug log uploads

https://streamable.com/ypingv

Expires in 2 Days

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-12-26:
@mpl1337 
sorry for the late reply, the video seems not available
could you share it again?

mpl1337, 2024-12-29:
> @mpl1337 sorry for the late reply, the video seems not available could you share it again?

https://vimeo.com/1042727398?share=copy#t=0

Haidiye00, 2024-12-30:
Hello, can you download ScreenToGif software ? Then record a GIF that flashes when opened by double clicking on the software? Can you provide us with the logs? You can find the logs based on this option
[image]
[image]
Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

closed 2025-04-17 by Haidiye00