bambulab/BambuStudio #5594: Bambulab studio keeps crashing

MaxHuiskes, 2024-12-21:
### Bambu Studio Version

v01.10.01.50-20241115162711

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

I5 gen 9th 16gb ram windows 10

### Printer

BambuLab P1S

### How to reproduce

1 Start computer.
2 Start Bambulav

### Actual results

bambulab studio throws error see image
[image]

### Expected results

Bambulab starts up

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MaxHuiskes, 2024-12-29:
i am getting this error 
[image]

lanewei120, 2024-12-30:
> i am getting this error ![afbeelding](https://private-user-images.githubusercontent.com/82757702/399172285-aab3dbdf-33b7-4771-803b-0de328f90f7d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1MjE2MTAsIm5iZiI6MTczNTUyMTMxMCwicGF0aCI6Ii84Mjc1NzcwMi8zOTkxNzIyODUtYWFiM2RiZGYtMzNiNy00NzcxLTgwM2ItMGRlMzI4ZjkwZjdkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMwVDAxMTUxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMyNDhlMTNlZmNkNzkyMTc3ZGY4OTE0NTliNDk0YjJjYjNjZjQwZjhhNzdmNzZlOWZkNTg0NmU4ODFmZTNkNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rFLm_kWkCdwZFNQ919dtABUtekEWejM40LCFEhyEmvQ)

could you upload the logs here?

following this guide
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

closed 2025-11-04 by MaxHuiskes