bambulab/BambuStudio #5596: Classic Wall Generator not setting the correct speed for some partial overhang walls

carribeiro, 2024-12-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Nome do dispositivo	DESKTOP-S0176R5
Processador	Intel(R) Core(TM) i5-4670 CPU @ 3.40GHz   3.40 GHz
RAM instalada	16,0 GB (utilizável: 14,9 GB)
ID do dispositivo	07C6AB48-53C0-4174-82F9-6F02D31C5020
ID do Produto	00330-81483-67934-AA740
Tipo de sistema	Sistema operacional de 64 bits, processador baseado em x64
Caneta e toque	Nenhuma entrada à caneta ou por toque disponível para este vídeo

### Printer

Bambu Lab A1 Combo 0.6m nozzle

### How to reproduce

1. Select a cone-shaped model with overhanging walls on the inside of the model (I'm printing a tent-like cover for a circus).
2. In my case the model is colored; I've not checked if the problem persists for non colored models but I'm assuming so.
3. Slice the model with the Arachne wall generator, and using the "speed" visualization, see that the internal overhanging walls are printed very slowly.
4. Now select the classic wall generator, and see that the resulting gcode tries to print the internal walls at the regular wall speed.

### Actual results

It's a long story but I promise that the bug report itself is pretty simple, but I'll try to explain how I got to this point.

It started when I tried to print a very shallow 12-sided prism, with 23° inclination, which is the tent-like cover of a caroussel prysm. Looking from the outside it isn't the kind of geometry that looks like have overhangs and such, as it seems to be entirely self-supported. However, whenever I tried to print it quality was poor. I kept adding more walls and top layers, but quality was still poor:

[image]

The samples above start from the lower left sample, and moving counterclockwise, I started with 2 walls and a very sparse internal structure, with bad results. I kept adding more walls and top layers. In the final sample I increased the infill density but quality was still poor.

[image]

Analysing the samples, I noticed that the innermost wall (which was being printed first) was overhanging, but it was not being printed at as bridge; it was printed at normal wall speed. Because of this the wall was dropping and did not support the upper layers, leading to the poor quality that was noticeable in the images above.

I could solve the problem by increasing walls, top layers, and infill even more,  but that goes again the spirit of the project. The tent cover should be as lightweight as possible. It makes no sense for it to be heavier than the base and structure of the carousel.

My assumption was that the slicer couldn't figure that this was a overhanging wall because it was supposed to be supported by the infill. But even an infill of 25% was still not enough.

In order to test this hypothesis I modified the model to transform it into a thin shell, and sliced it again. My assumption was that if the inside wall was unsupported the slicer would print it using bridge settings and the wall would print fine.

However, when I checked the slicer output, I noticed that with the classic slicer, the innermost wall (which is floating in space) was still being sliced like a regular wall. Slicing it in Arachne gave the correct result.

[image]

With Arachne the innermost wall prints at 10mm/s.

[image]

With the classic wall generator, the innermost wall is printed at more than 100mm/s.

### Expected results

The classic wall generator should be able to detect such overhanging walls, even if they are supported by the infill (as it was the case when I was slicing the prism), and specially when trying to print the shell.

### Project file & Debug log uploads

[Cobertura Carrossel 21-12 bico 06.zip](https://github.com/user-attachments/files/18219613/Cobertura.Carrossel.21-12.bico.06.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

carribeiro, 2025-01-04:
Is anyone lookin at this? This part is the hardest model to print that I've ever tried. Even with a lot of infill the result is always poor.

QingZhangBambu, 2025-01-27:
@carribeiro sorry to reply it later. you are right, the overhang detection is wrong. it should be fixed

QingZhangBambu, 2025-01-27:
[image]
I have fixed it and will release it after testing.

carribeiro, 2025-01-28:
Thanks for the reply AND FOR THE FIX! Don't worry about the delay, the most important thing is that it's fixed. Thanks!!!

QingZhangBambu, 2025-05-23:
@carribeiro The latest version works fine. Download it and have fun!

closed 2025-05-23 by QingZhangBambu