bambulab/BambuStudio #5599: Incorrect Password

chin-ming, 2024-12-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Device
2. Select Printer
3. Prompt Incorrect Password
4. Key In correct password
5. Idle for 15 minutes - 30 minutes ( computer lock screen)
6. Navigate out from Device
7. Navigate back to Device
8. Select Printer
9. Prompt Incorrect Password.
10.

### Actual results

Incorrect Password Prompt

### Expected results

No incorrect Password Prompt

In the BambuStudio.conf file
there are "access_code" & "user_access_code" fields : { "[printer serial]" : "[passcode]" }
"user_access_code"  -- passcode is the latest
"access_code"  --- password is incorrect.

to fix this:
Close Bambu Studio
manually update the "access_code" passcode 
Open Bambu Studio
Issue disappeared.

### Project file & Debug log uploads

[BambuStudio_debug.zip](https://github.com/user-attachments/files/18220640/BambuStudio_debug.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

chin-ming, 2024-12-22:
Can help to fix this issue on next Bambu Studio Release?

walterwongbbl, 2025-01-01:
@chin-ming 
We will fix and resolve this issue, and it has been added to the plan.

PearlBlueSoul, 2025-01-06:
Would like to add this is true on MacOS (ARM).   Recently cycled the access code on the printer, Bambustudio consistently erases the access code and claims it is in error.  Previous to cycling the access code it was retained with no issue.

arranf, 2025-02-05:
Also experiencing this on MacOS ARM. Putting in the same access code each time is a workaround.