bambulab/BambuStudio #5603: Allow "Change Type" on multiple parts at once

FatHammer, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
I sometimes want to change the types of multiple parts of an object at once (to negative, modifier, etc.), but having to do each one individually can get somewhat time-consuming.

**Describe the solution you'd like**
I would like to be able to select multiple parts at once, and change the type of them all at the same time.

**Describe alternatives you've considered**
Nothing really comes to mind.

**Additional context**
[image]

[image]

badkernel, 2024-12-26:
I agree, this is something that should be added as soon as possible.

vanowm, 2025-06-08:
Duplicate of [#1400](https://github.com/bambulab/BambuStudio/issues/1400)