bambulab/BambuStudio #5604: Missing dark mode

DankLab, 2024-12-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

P1S

### How to reproduce

Preferences. Line under media where dark mode select should be is missing.
[image]

### Actual results

Bambu studio works fine just eye piercing with white theme.

### Expected results

I expect the program to have a dark mode once more. 

### Project file & Debug log uploads

No need since this is a feature that's missing in bambu studio.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-12-23:
Ubuntu's dark mode follows the system settings and cannot currently be specified independently.

[image]

DankLab, 2024-12-23:
That's even better but it doesn't work for me, any suggestions on what I can do to fix it? Only the top of the window changes when i switch theme. 
[image]
[image]

namdnas, 2024-12-24:
I believe what you want to find might be `~/.config/bambu-studio/settings.json` on linux. I have it installed on Windows and the location is `C:\Users\%username%\AppData\Roaming\BambuStudio\BambuStudio.conf` and the line you would want to update is `"dark_color_mode": "<>",` It will probably initially be set as 0 set it to 1 instead.

closed 2025-03-31 by MackBambu