bambulab/BambuStudio #5605: Layers skipped

OmiCron07, 2024-12-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Use my 3mf file and print the plate 9.

### Actual results

When sliced, layer #31 shows a full support layer in the preview, just before the support interface, but when printing, it never did this support layer.

Layer #30:
[image]

Layer #31:
[image]

Layer #32:
[image]

### Expected results

It should not skip layers or show false previews.

### Project file & Debug log uploads

[Ultimate Mini Nintendo NES Cartridge Keychain Pack.zip](https://github.com/user-attachments/files/18225642/Ultimate.Mini.Nintendo.NES.Cartridge.Keychain.Pack.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

NothFie, 2024-12-26:
@OmiCron07 Thanks for your reporting. Can I have a look at the model you printed? We reproduced it without changing any setting and found layer_31 did exist(though the lines are not connected well).
[image]

OmiCron07, 2025-01-12:
@NothFie This is exactly what happened to me too.

NothFie, 2025-01-13:
@OmiCron07 I gei it. In this situation, I suggest you reduce the "Base pattern spacing" in support settings. Because the lines are so thin that layer_31 is not stable if the spacing are too big.

Hilo4321, 2025-02-09:
All of my prints are turning out similarly, which causes my support for PLA/PETG material to fail and sink into the support structure, and make removing supports very difficult. Even with a very basic file like the one below will have this happen.

https://makerworld.com/en/models/242830?from=search#profileId-803646

See pics here, I can't share the files for the print with the pictures, but the same symptoms are displayed. The base support (PLA) doesn't cover the area sufficiently. When the support material goes onto that layer, it's falling through/not bridging correctly. I even took a screenshot of the slicer where I expected there to be a full layer of PLA.

This is totally broken and makes large flat areas impossible to print effectively. Very frustrating because I'm wasting material that costs $70/kg! This was supposed to make things nicer, but instead makes them look TERRIBLE.

https://photos.app.goo.gl/GuxE8sZZWqGcsGQ29

Hilo4321, 2025-02-10:
I appreciate the response. Looks like a success so far using the test print I linked previously.

The only changes I made were:

layer height: .12mm to .20mm
base pattern spacing: 2.5mm to 2mm

I look forward to being able to set those few layers separately, so I can print more finely with supports.

closed 2025-02-17 by NothFie