bambulab/BambuStudio #5610: Color Painting by Height results in more than one Filament change times more than 1

locou, 2024-12-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Load an object with variying top layer heights.
2. Use the Color Painting tool with the Tool type set to "Height Range"
3. Paint in some top layers

[image]

### Actual results

Filament is changed multiple times for internal bridges. For this project its "5".

[image]

### Expected results

Filament change times should be "1" since the coloring is done by height. I don't want internal color bleed for bridges.

### Project file & Debug log uploads

[bambucolorchangebug.zip](https://github.com/user-attachments/files/18237304/bambucolorchangebug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

locou, 2024-12-31:
here's another 3mf file which reports in 4 filament changes, but I cannot figure out why and where they are performed.
[unnessesary_filamentchanges.zip](https://github.com/user-attachments/files/18282967/unnessesary_filamentchanges.zip)
[image]

zhimin-zeng-bambulab, 2025-01-07:
[image]
Currently, the depth of color penetration cannot be set. 
You can clear the color of the object first, and then set the custom gcode to change the filament, as shown in the picture above.