bambulab/BambuStudio #5612: Minor issue for Ubuntu Linux and video playback

buelowp, 2024-12-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

- 8 × Intel® Core™ i7-7700K CPU @ 4.20GHz
- 62.7 GiB of RAM
- 6.8.0-51-generic (64-bit)
- NVIDIA GeForce GTX 1080/PCIe/SSE2

### Printer

X1Carbon

### How to reproduce

Download 1.10.1.50 for ubuntu 24.04, set executable permissions, and run the application from the command line. 

`# ./Bambu_Studio_linux_ubuntu_24.04_v01.10.01.50.AppImage`

I always run the first time to make sure it's going to start without extra appimage work.

### Actual results

The application works as expected, mostly, but the debug output is noisy with the following lines.

```
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e3e0c0] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x763977e4f080] deprecated pixel format used, make sure you did set range correctly
```

### Expected results

This is mentioned in the following post
https://superuser.com/questions/1273920/deprecated-pixel-format-used-make-sure-you-did-set-range-correctly
and can likely be resolved by calling ffmpeg with quieter output format options.

Not spamming the console with this single output makes it easier to find other issues should the arise.

### Project file & Debug log uploads

[ground signal base_fixed.zip](https://github.com/user-attachments/files/18241217/ground.signal.base_fixed.zip)
[log.zip](https://github.com/user-attachments/files/18241237/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-12-26:
Thanks, let's study it.

nicocesar, 2025-01-01:
I have the same issue and the image is all pixelated
[image]