bambulab/BambuStudio #5614: Error when exporting G-Code file

imba-LALA, 2024-12-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11 23H2

### Printer

Anycubic Kobra Plus 0.4 nozzle

### How to reproduce

1.Import Model
2.Slice Plate
3.Export G-code file

### Actual results

The pop-up window displays:
Failed processing of the filename_format template.
Parsing error at line 1: Not a variable name
{input_filename_base}_{filament_type[initial_tool]}_{print_time}.gcode

![Bug](https://github.com/user-attachments/

### Expected results

Output G-code file.

### Project file & Debug log uploads

This problem occurs regardless of the model used.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

imba-LALA, 2024-12-25:
[image]

imba-LALA, 2024-12-25:
[image]

imba-LALA, 2024-12-25:
[image]

kkeonline, 2025-01-09:
I have submit a pull request 
https://github.com/bambulab/BambuStudio/pull/5734

XunZhangBambu, 2025-06-03:
@kkeonline  This has been fixed in our latest verison.