bambulab/BambuStudio #5615: Bambu Studio CTDs

worvik, 2024-12-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 10.0.22631 Build 22361

### Additional system information

Intel i9-14900K
128gb RAM
NVIDIA GeForce RTX 3090

### Printer

Bambu Lab X1C

### How to reproduce

Slicing Project Bambu Studio CTDs

### Actual results

Slicing Project Bambu Studio CTDs

### Expected results

Complete slicing

### Project file & Debug log uploads

[crash_Wed_Dec_25_10_32_47_0.zip](https://github.com/user-attachments/files/18246911/crash_Wed_Dec_25_10_32_47_0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2025-01-24:
Could you provide the 3mf?

worvik, 2025-01-24:
I'm not able to open the step file without it crashing so no I can't save it as a 3mf

[knurled+knob+with+ring.zip](https://github.com/user-attachments/files/18534614/knurled%2Bknob%2Bwith%2Bring.zip)

QingZhangBambu, 2025-01-27:
@worvik Ok， could you upload dump files? i will ask concerned staff to check this.

MackBambu, 2025-02-05:
hi,@worvik This STEP model looks a bit strange; it seems to be composed of triangular faces. If you have the original STL or OBJ file, you can slice it directly. Converting it back to a STEP file has no benefits and will only multiply the import time.
[image]

closed 2025-08-30 by worvik