bambulab/BambuStudio #5616: A1 mini connection to Iphone app, but shows offline in babu studio. It have been working for months, but just disconnected yesterday. The app still works and the printer is connected to internett

Amalmei, 2024-12-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab A1 mini

### How to reproduce

Have tried to log out, factory reset printer, change wifi network.

### Actual results

Printer shows up in bambu stuio, but its not connecting. the app works as normal where i can start prints.

### Expected results

I was hoping it would connect to studio.

### Project file & Debug log uploads

[crash_Wed_Dec_25_19_49_34_0_done.log](https://github.com/user-attachments/files/18247033/crash_Wed_Dec_25_19_49_34_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

eValker, 2024-12-25:
I had exactly the same issue. It seems to be related to this report: https://github.com/bambulab/BambuStudio/issues/5436. For me, the solution was to set all filaments to generic PLA.

danifet, 2024-12-25:
I have tried all the possible instructions to fix it. One of the issues is that both Test Storage Upload and Test Storage Upgrade failed in the Network Test. Changing the filament to PLA didn't help.

Amalmei, 2024-12-26:
> I had exactly the same issue. It seems to be related to this report: #5436. For me, the solution was to set all filaments to generic PLA.

I changed all to generic PLA and its now online in bambu studio again. Thank you so much for the solution!

DanBao-Bambu, 2025-01-13:
We are investigating this issue and can try to avoid it by creating a new custom filament, deleting the problematic one (the consumable that cannot be connected to the current printer after setting it up), and checking if the problem still exists.