bambulab/BambuStudio #5617: No matter what I try, when I try to Open Bambu, it opens, but there is no Window. So it looks like it's opened, and my Macbook shows me that it is opened, but there is no Window. I even completely deleted Bambu Studio from my MacBook, but it still doesn't show a Window. What is going on ?

janwaldicloudcomm, 2024-12-26:
### Bambu Studio Version

10.15

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Seqouia

### Printer

I don't know, because I can't see it

### How to reproduce

1. Download Bambu Studio
2. Open it 
3. connect a second display
4. close the MacBook
5. try to open Bambu Studio again
6. realise that it won't open

### Actual results

On the Macbook it looks like it is open, but there is no Window to see, even, when you delete the Bambu Studio, and reinstall it

### Expected results

The Application should open and show a Window

### Project file & Debug log uploads

[Bug Report.3mf.zip](https://github.com/user-attachments/files/18247600/Bug.Report.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-12-30:
If the second display is not connected, can Studio start normally?

davidhasek1, 2025-11-13:
@janwaldicloudcomm 
Thank for a tip. I tried what you said... For some reason I see just a rim around a transparent window (upon a description I suggest that is a window). That's a weird behavior.
[image]

When I tried to arrange windows side by side I see blank black window without chance to close it or do something meaningful.

[image]

Also I see Bambulab studio top bar options without actual window

[image]

Macbook may cause this problem, but It looks like Bambu studio is not optimized from some version of macOS.. I am just guessing because I noticed this issue with previous version of macOS

Thank you for your response!