bambulab/BambuStudio #5618: "Flush into object" and "flush into infill" not available if prime tower is disabled

tangcla, 2024-12-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1 C

### How to reproduce

Right-click object and flush into object should be visible (see screenshot)
[image]

### Actual results

With prime tower disabled, the options are greyed out (and the preview shows no flush into object)
[image]
[image]

### Expected results

[image]
[image]

### Project file & Debug log uploads

[l
[UNO.zip](https://github.com/user-attachments/files/18249898/UNO.zip)
og.zip](https://github.com/user-attachments/files/18249891/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

XunZhangBambu, 2024-12-29:
@tangcla  Thank you for your feedback. We will discuss whether to remove this restriction

EagleoutIce, 2025-02-15:
Is there any update on the results of this discussion? :)

Schogette-Xtra, 2025-04-14:
> [@tangcla](https://github.com/tangcla) Thank you for your feedback. We will discuss whether to remove this restriction

One more for "please remove the restriction". Thank you!

OrigamiMarie, 2025-07-01:
Same here, I would like to make random duckies that absorb all / most of the spare filament, instead of it turning into useless towers.

samster395, 2025-07-18:
Any update on this?

I see no reason for there to be restriction on this.