bambulab/BambuStudio #5621: Constantly losing connection with the printer

nghtstr, 2024-12-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.2 Sequoia

### Additional system information

M3 Pro
18GB of Ram

### Printer

Bambu Lab A1

### How to reproduce

After a short period of time, Bambu Lab Studio will not be able to find the printers. This causes me to have to re-login. The application does not crash nor is it exited normally. It will randomly just stop being able to talk to the Bambu Lab servers to communicate with my printers.

### Actual results

As stated above, the Bambu Lab Studio application becomes unable to communicate with the Bambu Lab Servers to communicate with my printers.

### Expected results

I should not have to re-login every other day (or sooner) to print items from Bambu Lab Studio.

### Project file & Debug log uploads

I do not think this is necessary for a Network related error.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

skintigh, 2024-12-26:
Does the exact same for me in Windows, except up to 50 times per day when in LAN-only mode. Even switching tabs is enough to force me to re-type my password. 10, 15 times per print I have to retype my password. Every time the laptop goes to sleep, have to retype my password.

This is intentional. Rather than re-trying a connection that seems interrupted or dropped, like every other piece of software on Earth, instead BS gives up after 0 tries and then deletes all the passwords, because f*ck users.

This appears to be their way to ram their cloud service down your throat.

walterwongbbl, 2025-01-01:
@nghtstr 
Does this happen in LAN mode? Can you provide the log file?

nghtstr, 2025-01-02:
I do not know how to use LAN mode. Since all of my devices are connected via WiFI, I thought the only course of action I could use was the cloud service. If this is inaccurate, please instruct me on how to do LAN mode via WiFi.