bambulab/BambuStudio #5622: Constant FTP errors in LAN-only mode

skintigh, 2024-12-26:
### Bambu Studio Version

There is no About screen with that info, but the latest

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1C

### How to reproduce

Print

### Actual results

Either 30% or 70% progress then "FTP error" nearly 100% of the time when on LAN-only mode.

The only way to print usually is to turn off the firewall, even though I have rules allowing the app through the firewall. It seems your printer firmware freaks out and rebroadcasts packets repeatedly, breaks and restarts connections, and other insanity that results in packets being dropped nearly 100% of prints.

It also seems your firmware and software are intentionally written to be anti-user and drop connections on purpose without retrying even once, and then deleting passwords for no reason, forcing me to retype the password up to 50 times per day, 10-15 times per long print to see progress.

It also seems your fake "FTP errors" are related to filenames. I named a file "ring0.stl" and it couldn't print. I renamed it "ringzero.stl" and it could print, with the firewall still on. Yet tests of removing numbers from other filenames still resulted in "FTP errors."

### Expected results

Print

### Project file & Debug log uploads

It appears to be the printer firmware with the problem, not my project, unless 100% of my projects and 100% of downloaded projects all magically have the same errors.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

thurcombe, 2025-05-11:
I recently experienced similar issues after many many hours of rock solid LAN only printing, sounds similar. Either FTP connection errors and/or reauth requests that would also fail. I believe this is due to I/O errors, replacing the SD card resolved my issue.

walterwongbbl, 2025-05-23:
Hi @thurcombe 
What is your printer model

walterwongbbl, 2025-05-23:
If you have new questions, please create a new reply thread.

closed 2025-05-23 by walterwongbbl

thurcombe, 2025-05-23:
@walterwongbbl also an X1C

skintigh, 2025-05-24:
I'm actually the one who opened this issue. For me it seems a bad Intel
wifi driver was interfering with something, but *only *when the firewall
was on, making it look like a firewall or WFP issue. It went away when I
upgraded a week or so ago.

Bambu Studio reacted to this interference in 2 ways:
1) False errors about FTP failures even though the FTP transfer was
successful
2) Hangs at 70% and hard crashes. I've tried 6 versions of BS now, they all
crash the same way on fake FTP errors: first they hang for 5 minutes, and
if you press "cancel" during that time BS crashes and must be restarted.
Every time. Every version. After the first hang at 70%, subsequent hangs
would occur at 30%.