bambulab/BambuStudio #5625: Text layer appears lower and in different color than assigned, messing up my ability to print properly

nickth0, 2024-12-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.7.1

### Printer

X1C

### How to reproduce

Add text on to a layer
Different color bottom layer and different color text layer
Text appears below the surface in one color, disappears, then appears above the surface in another color.

### Actual results

Video was too big for GitHub, so posted it on YouTube:
[https://youtu.be/urlgZ5DuprY?si=p4hZiZv8jxmgFVN3](url)

Also adding a screenshot of the text settings here.
[image]

### Expected results

The layers should simply be green then white, with the text starting on the white layer, not below it. The text is meant to be white not green. The text shouldn't start below the white top surface layer. The green layers are meant to go higher up as I painted them that way.

### Project file & Debug log uploads

[Outlet-plate-Bambu-bug-report.3mf.zip](https://github.com/user-attachments/files/18257416/Outlet-plate-Bambu-bug-report.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2025-02-05:
The internal algorithm of the software causes the white layer to penetrate several layers downward. This is to prevent the white layer from being too thin and causing the paint inside to penetrate into the outer surface.

nickth0, 2025-02-05:
Please fix the user experience to work as expected, thank you!