bambulab/BambuStudio #5626: Print cancelled mid-print

tmachine3d, 2024-12-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

CPU: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz
RAM: 16.0 GB

Printer firmware: 1.07.00.00
SN: 01P09A452803915

### Printer

Bambu Lab P1S

### How to reproduce

Print plate
Single-filament print, multiple objects

### Actual results

Failed 50% of the way through. Extruder head stopped mid pass. Printer reported cancelled print. No reason or justification.  No power outage, no triggering event.  Picture of extruder head position and print status attached.

[image]

### Expected results

Print should have completed automatically.

### Project file & Debug log uploads

Project file: [JiaoDragon+Egg.zip](https://github.com/user-attachments/files/18257394/JiaoDragon%2BEgg.zip)
Log files: [debug_Wed_Dec_25_13_38_59_10180.log.zip](https://github.com/user-attachments/files/18257421/debug_Wed_Dec_25_13_38_59_10180.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tmachine3d, 2024-12-27:
Since I want to continue printing, I have seen posts where downgrading firmware to the previous version helped, so I did that this evening. After this print cancellation issue, I downgraded to Printer firmware: 1.06.01.02, with AMS firmware 00.00.06.44.  I will notify here if the same problem persists.

tmachine3d, 2024-12-27:
Also, since it is mentioned in the forums, I am just using the stock SD card and have not used any other since acquiring the machine, sending jobs only wirelessly from my laptop or phone.

tmachine3d, 2024-12-27:
Made it through the print with the downgraded firmware. Not sure if that is a cause since it was just one failure, one success, but didn't want to waste more filament. This is the forum thread where I have seen many other comments about this problem:  https://forum.bambulab.com/t/print-cancelled-but-why-and-by-what/118998/23

gaolegao-lx, 2025-01-02:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

tmachine3d, 2025-01-05:
I am unable to upload those logs through the Handy app because they have
rolled off.  The issue took place on Dec 26th. The zipped logs were
included in the Github support ticket submission though.

tmachine3d, 2025-02-17:
UPDATE: I have not had a recurrence if this problem through many prints since downgrading my firmware back on 12/27 and have not upgraded firmware again.

bmod, 2026-05-02:
My print was just cancelled mid-print for no reason and Bambu Handy tells me I did it.

I did not cancel the print, the app lied. Maybe the app sent a cancel command while opening?

Regardless, this is unacceptable, no such thing should ever happen.