bambulab/BambuStudio #5631: Text shape's "Object coordinates" are miscalculated on X and Y

ShaneQi, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.6.1 (23G93)

### Printer

X1C

### How to reproduce

1. New BambuStudio project
2. Add a cube primitive on plate (default size is fine)
3. Add a "text shape" with **size 8** and **text "M3x22"** (the size and text are irrelevant to the issue, but this size and text is easy to spot the issue)
4. Move the "text shape" part, and change to "Object coordinates" if not already. Update X value and Y value to 0.
5. The text SHOULD be centered on X and Y axis with the cube, but it's NOT.
6. If change the view angle to "top", you can also see that when the "Object coordinates" of the text shape is X0Y0, it's NOT centered on X or Y axis with the cube. 
7. It seems like that the "Object coordinates" of text shapes are mis-calculated.

### Actual results

When a text shape's "Object coordinates" is set to X0Y0, it's NOT centered on X and Y axis with the main object.

### Expected results

When a text shape's "Object coordinates" is set to X0Y0, it should be centered on X and Y axis with the main object.

### Project file & Debug log uploads

[text-shape-object-coordinates-issue.3mf.zip](https://github.com/user-attachments/files/18265616/text-shape-object-coordinates-issue.3mf.zip)
[log.zip](https://github.com/user-attachments/files/18265617/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-12-30:
Hello, our internal version has identified this issue and is currently fixing it~

platini76, 2025-01-06:
when???

Haidiye00, 2025-01-06:
Hello, Sorry, the feature has not been fixed yet, and I cannot confirm the release time of the version~

platini76, 2025-01-06:
is there a solution to bypass it for now?

Haidiye00, 2025-01-06:
hello,You can use version 1.9 to edit it, then open it with the latest version and use the translation tool to translate it~

platini76, 2025-01-06:
1.9 cannot modify a file that is saved with 1.10 version

Haidiye00, 2025-01-06:
hello, use version 1.9 to generate text ,and use version 1.10 to move text position~

platini76, 2025-01-06:
thanks, but it is a bad bug... and 2 months old.. bambu must wake up..

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00