bambulab/BambuStudio #5632: AMS secondary filament drop down not showing all listed filaments

Bandy1284, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

When attempting to change filaments in the AMS. 
Open Studio, Device, select filament slot, generic PLA, PA profile (Drop Down) you will see here it only lists the first 10 filaments and you can not scroll down to select any other filament.
[image]
 

### Actual results

You get a list of the first 10 filaments,  their is no scroll bar to go further down and select any other

### Expected results

I should be able to select any of my saved filaments. 

this happens in any scenario not specifically on one project. 

### Project file & Debug log uploads

[Project file to uplaod.zip](https://github.com/user-attachments/files/18265782/Project.file.to.uplaod.zip)
[debug_Thu_Dec_26_20_46_41_3932.log.zip](https://github.com/user-attachments/files/18265790/debug_Thu_Dec_26_20_46_41_3932.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Bandy1284, 2025-01-05:
An update on this would be nice as id like to be able to fully use my AMS

walterwongbbl, 2025-01-16:
Hi @Bandy1284 
Are you saying that you can't scroll the list? Can you do it using the mouse wheel?

Bandy1284, 2025-01-16:
> Hi [@Bandy1284](https://github.com/Bandy1284) Are you saying that you can't scroll the list? Can you do it using the mouse wheel?

Nope can't scroll that list at all, All it will show is the first 10 filaments. The mouse wheel does not scroll it. 

The filament type list works... but when trying pick my custom filaments that I have tuned I only get the first 10.

Bandy1284, 2025-01-27:
Any update at  all, or can you yourself scroll the list?