bambulab/BambuStudio #5635: Calibration

Spravko, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz   2.19 GHz; ram 16GB

### Printer

X1 carbon

### How to reproduce

Every time I switch to calibration, I have to select the printer.

[image]

### Actual results

Even if I select the printer, the flow dynamics calibration cannot be started.

[image]

### Expected results

In older versions of bambu studio there was no problem with this. It works without problems in Orca slicer.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18266251/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2025-02-05:
It has been fixed in v1.10.2 Public Beta.

closed 2025-02-05 by zhimin-zeng-bambulab

Spravko, 2025-02-05:
👍