bambulab/BambuStudio #5637: AMS Auto Refill pulled from incorrect AMS slot

Crowsnest3D, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab P1S

### How to reproduce

Had the following AMS slots used for this project:
A1) PLA Basic Blue
A2) PLA Basic Black (Almost empty spool)
A3) PLA Basic Black (Full Spool)
A4) PLA Basic Green
B5) PLA Basic Red
(Slots 6-8 irrelevant here)

When sending the print to the printer, I selected AMS slots 1/2/4/5.

### Actual results

According to the Auto Refill window the arrows implied that "A2" should have restocked to "A3":
[image]

However instead mid-print when A2 ran out, it used A1 to restock. This being an entirely different color ruining the print.

### Expected results

The AMS should follow the order provided in the UI (Direction of arrows) as otherwise it is an unituitive user experience.
As well, more than just material choice should be used to determine which slot is refilled - if another of the same color is stocked, anything other than that is generally undesirable.

### Project file & Debug log uploads

[Project & Log.zip](https://github.com/user-attachments/files/18266378/Project.Log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Leeshha, 2025-01-06:
I am also seeing the same problem.  AMS had All Bambu PLA 

- A1 Blue 
- A2 Grey 
- A3 Black 
- A4 White. 
When A3 (Black) ran out it decided to just switch over to A1 and continue the print with Blue
[image]

upirx, 2025-03-28:
Similar problems remain in Bambu Studio 2.0.0.95, it treats Bambu PLA Matte Charcoal (at AMS C3 slot) and Bambu PLA Matte Marine Blue (at AMS A2 slot) as the same material.

[image]