bambulab/BambuStudio #5642: Player is malfunctioning [Mac]

mrkevinham, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOs Sierra 15.2

### Printer

Bambu Labs P1S

### How to reproduce

1. Go to Devices
2. Click play button on player

### Actual results

Player stays black.
Receive error: Player is malfunctioning. Please reinstall the system player.

Occurs with any print.
Bambu Studio app on my iPhone shows the live preview without issue

### Expected results

Live view of print in process

### Project file & Debug log uploads

[debug_Sat_Dec_28_12_21_49_29875.log.0.zip](https://github.com/user-attachments/files/18267274/debug_Sat_Dec_28_12_21_49_29875.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

AndreasFeldt, 2024-12-29:
I have the same issue, have also tried reinstalling bambu studio without success

bambu123, 2024-12-30:
Same As #5383

gcstr, 2024-12-30:
The same thing started happening with my yesterday right after the firmware update on the printer
[image]

c0desurfer, 2024-12-30:
Same problem on Mac with X1C.

lennvilardi, 2024-12-30:
Same here too with X1C

c0desurfer, 2024-12-30:
Interesting stuff in log files.

```
2024-12-30 12:42:35.394081[Thread 0x00000001efd44240]:MediaPlayCtrl::TogglePlay
2024-12-30 12:42:35.396667[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Initializing... tutk_state: disable
2024-12-30 12:42:36.090895[Thread 0x00000004b59a3000]:MediaPlayCtrl: bambu:///tutk?uid=JDCV************111A&authkey=********&passwd=******&region=us&device=<SNIP>&net_ver=01.10.01.01&dev_ver=01.08.02.00&refresh_url=1&cli_id=efd22cc1-46cd-4b85-aca6-0036cc6ad511&cli_ver=01.10.01.50
2024-12-30 12:42:36.091348[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Loading... tutk_state: disable
2024-12-30 12:42:36.092417[Thread 0x000000016bccb000]:MediaPlayCtrl: start load
2024-12-30 12:42:36.096677[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] IOTC_Connect_ByUIDEx(JDCV************111A).......
2024-12-30 12:42:36.220987[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] IOTC_Connect_ByUIDEx OK SID [0]
2024-12-30 12:42:36.221009[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] Device[JDCV************111A] IOTCVersion[4030302]
2024-12-30 12:42:36.221013[Thread 0x00000001efd44240]:[BambuTunnelTutk::dump_stat] Device is from 192.*****.168:35378[JDCV************111A] Mode=LAN
2024-12-30 12:42:36.221016[Thread 0x00000001efd44240]:[BambuTunnelTutk::dump_stat] Device info VER[4030302] NAT[0]
2024-12-30 12:42:36.221735[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] avClientStartEx failed [-22005]
2024-12-30 12:42:36.221745[Thread 0x00000001efd44240]:[BambuTunnelTutk::tutkClose] IOTC_Session_Close calling [0]....
2024-12-30 12:42:36.221784[Thread 0x000000016bccb000]:MediaPlayCtrl: end load
2024-12-30 12:42:36.222313[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2024-12-30 12:42:36.222333[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: size: 0x0
2024-12-30 12:42:36.222509[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Please check the network and try again. You can restart or update the printer if the issue persists. [2:-22005] <12-30 12:42> tutk_state: disable
2024-12-30 12:42:36.223526[Thread 0x000000016bccb000]:MediaPlayCtrl: start stop
2024-12-30 12:42:36.223538[Thread 0x000000016bccb000]:MediaPlayCtrl: end stop
2024-12-30 12:42:36.223606[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2024-12-30 12:42:36.223743[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2024-12-30 12:42:36.228042[Thread 0x00000001efd44240]:[BambuTunnelTutk::tutkClose] IOTC_Session_Close OK
2024-12-30 12:42:36.228060[Thread 0x00000001efd44240]:[BambuTunnel::close]
2024-12-30 12:42:41.474646[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Initializing... tutk_state: enable
2024-12-30 12:42:42.097027[Thread 0x00000004b59a3000]:MediaPlayCtrl: bambu:///tutk?uid=JDCV************111A&authkey=********&passwd=******&region=us&device=<SNIP>&net_ver=01.10.01.01&dev_ver=01.08.02.00&refresh_url=1&cli_id=efd22cc1-46cd-4b85-aca6-0036cc6ad511&cli_ver=01.10.01.50
2024-12-30 12:42:42.097258[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Loading... tutk_state: enable
2024-12-30 12:42:42.099213[Thread 0x000000016bccb000]:MediaPlayCtrl: start load
2024-12-30 12:42:42.105539[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] IOTC_Connect_ByUIDEx(JDCV************111A).......
2024-12-30 12:42:42.263853[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] IOTC_Connect_ByUIDEx OK SID [0]
2024-12-30 12:42:42.263901[Thread 0x00000001efd44240]:[BambuTunnelTutk::open] Device[JDCV************111A] IOTCVersion[4030302]
2024-12-30 12:42:42.263909[Thread 0x00000001efd44240]:[BambuTunnelTutk::dump_stat] Device is from 192.*****.168:35378[JDCV************111A] Mode=LAN
2024-12-30 12:42:42.263917[Thread 0x00000001efd44240]:[BambuTunnelTutk::dump_stat] Device info VER[4030302] NAT[0]
2024-12-30 12:42:42.265459[Thread 0x000000016bccb000]:MediaPlayCtrl: end load
2024-12-30 12:42:42.269885[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2024-12-30 12:42:42.269917[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: size: 0x0
2024-12-30 12:42:42.270413[Thread 0x00000001efd44240]:MediaPlayCtrl::SetStatus: Player is malfunctioning. Please reinstall the system player. [2:2] <12-30 12:42> tutk_state: enable
2024-12-30 12:42:42.271654[Thread 0x000000016bccb000]:MediaPlayCtrl: start stop
2024-12-30 12:42:42.271669[Thread 0x00000001efd44240]:MediaPlayCtrl::onStateChanged: 0, last_state: 3
2024-12-30 12:42:42.271675[Thread 0x000000016bccb000]:MediaPlayCtrl: end stop
```

JeremyLaurenson, 2025-01-01:
I am having the same issue.

m67hoff, 2025-01-01:
I have the same issue - the problem started with the MacOS update to 15.2    before it was working.    (BTW:  same issue in Orca slicer)

sneakyjoeru, 2025-01-01:
Same, adfter update

bobg2023, 2025-01-01:
I have the same issue

mkpansare, 2025-01-06:
Same issue, can not disable firewall or update the settings of firewall since Mac belongs to the company. Need another solution.

JeremyLaurenson, 2025-01-06:
Check that this is not an issues with **Little Snitch** if you have it installed.

c0desurfer, 2025-01-06:
> Check that this is not an issues with **Little Snitch** if you have it installed.

I hoped it would be something like that, I use `Little Snitch` myself. However Bambu Lab already acknowledged this issue in a support ticket I opened with them.

> We have indeed noticed and tested that the MacOS Sequoia 15.2 update has caused Bambu Studio to be unable to access video properly.

mkpansare, 2025-01-06:
Oh good. Thats at least good to hear that its on their radar. By any chance did they provide any ETA for a fix ?

c0desurfer, 2025-01-06:
> Oh good. Thats at least good to hear that its on their radar. By any chance did they provide any ETA for a fix ?

Not really.

> We are working diligently to resolve this issue and will do our best to fix it as quickly as possible.

SC-JBG, 2025-01-08:
I am having the same issue and could mitigate it by turning off the MacOS internal firewall. Once turned off (under Settings > Network > Firewall > Toggle OFF), the video feed works as expected. 
It does not help to create a rule in the firewall options to allow BambuStudio for incoming network connections - you must turn off the firewall entirely. 
This, of course, is just a workaround and no real solution, but I hope it helps pinpoint the actual issue and resolve it.

c0desurfer, 2025-01-09:
> I hoped it would be something like that, I use `Little Snitch` myself. However Bambu Lab already acknowledged this issue in a support ticket I opened with them.

Bambu Lab asked me again to disable Little Snitch today and after doing it I was able to connect to the camera successfully. Strange because I have no rules which would block Bambu Lab Studio.

SC-JBG, 2025-01-09:
> Check that this is not an issues with Little Snitch if you have it installed.

Thanks for the reply! I do not have Little Snitch installed on my machine. I can trace it back to the internal firewall being enabled or not.

JeremyLaurenson, 2025-01-09:
Yes, agreed it's an issue. My org downs allow me to turn off the Firewall... 
[image]
So I am in the same boat.

acircleda, 2025-01-18:
> I am having the same issue. Company enforced policy does not allow me to turn off the firewall (I wouldn't anyway) and this was not an issue in the past.

Exact same situation. Started after 15.2 OS update. Cannot change company policies. Camera was not a issue beforehand.

mkpansare, 2025-02-05:
Updated to MacOS 15.3 and now its working. Maybe it was an issue on Apple's side ?

garglebutt, 2025-02-05:
Yes, I found it started working after upgrading to 15.3. Whatever the incompatibility was it’s gone away now thank goodness.Peter Enzerink

mrkevinham, 2025-02-17:
Camera is resolved with os 15.3

iRaccoon13, 2025-03-03:
I'm running 15.3 and am still having this issue with a1. I've tried the latest Bambu Studio and Orca Slicer. Also, the SD Card files won't load. I've turned off my firewall. All other functionality seems fine

freeyoung, 2025-03-26:
Running 15.3 as well for quite some time and hitting the issue all of a sudden. Interestingly, the issue does not exist on my other mac.