bambulab/BambuStudio #5644: Stringing bug report + solution or workaround

newoldschool, 2024-12-28:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS sequoia

### Additional system information

Hello I want to report a bug that I was suffering for severals days or weeks I'm not sure for how long.

I've lost a crazy amount of hours searching for solution on internet, trying several settings at every part of the slicer software, and more thank 2kg of filaments wasted on test prints and lost prints for clients due to this issue.

I was suffering of very bad stringing, I knew my settings were at least okay, everything should we working all right.
I tried everything from temperature to retraction length and speed, z-hop, checking a lot of checkboxes I tried everything no kidding.

Until I accidentally changed ONE little setting because I suddenly put 4 models in my plate to test different this...
I've just changed the option "Print sequence: By layer" to "By object".

Just that. And suddenly all the stringing disappeared.
Using the same settings, I even tried other similar settings in retraction and z-hop etc just to see if was a lucky shot, but changing settings was not an issue, the stringing always appear when I select By laver, and the are always gone when I select By Object.
After that I tried with fresh new files, from zero, and different testing models and always the same workaround.

Just to clarify, maybe this was happening before, sometimes I attributed the problem to wet filament, or difficult geometry, I used workarounds or sometimes I just burn  the strings with the heatgun. But right now was being really bad stringing, attached you can find to examples, one is the test model, side by side with both printer sequence settings, and the other is an example of a print for a client.

Maybe this is just the tip of the iceberg, and with this you can start to solve that bug.
I really hope you can find a solution for this very soon and give us a new software update.
I've lost so many hours not working and testing settings, and so many filament, the best news I can receive is that you found the solution and you can give us a software update.

Thanks in advance!

### Printer

Bambu Lab A1

### How to reproduce

1. import a stringing test like the one I've uploaded
2. be sure you have select By layer in the plate settings, and do not touch the Process settings, do not change the Print sequence there, just in the plate settings.
3. print
4. after the print is finished, change the plate settings now choose the option By object.
5. print

### Actual results

With "by layer" option selected, there is stringing, sometimes is not that much, in a lot of cases (in my case) is really bad.
With "by object" option selected, no stringing at all and great quality printing.

### Expected results

"by layer" and "by object" should give the same exact quality print result.

### Project file & Debug log uploads

[Stringing bug report + solution or workaround.zip](https://github.com/user-attachments/files/18267884/Stringing.bug.report.%2B.solution.or.workaround.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-12-30:
[image]
are you print 2 of these with by object sequence?

newoldschool, 2024-12-30:
Hello, no, just one.
What is happening to me is that even if I have just one in my plate, selecting in the plate settings "by object" will print it without any stringing, I can make a video if you want, it's so amazing and strange and the same time to see how setting like this con solve the stringing problem (or maybe we can think the other way and say the selecting the default "by layer" is causing stringing for some reason, because I don't touch any other settings, is like I can turn "stringing" on and off with the "by object" check box)

QingZhangBambu, 2024-12-31:
Yeah, that's amazing. There should be something wrong. I will try it.

ziehmon, 2025-01-03:
> Yeah, that's amazing. There should be something wrong. I will try it.

@XunZhangBambu this is the issue discussed/solved in https://github.com/bambulab/BambuStudio/pull/4848. 

When print sequence is set to "by object", timelapse gcode will not be injected and the Z hop type will not be set to Normal. I am testing the fix discussed in this PR and you can expect the new PR in the next days.

newoldschool, 2025-01-05:
Hello thanks for the info @ziehmon , what is PR? I'm not very fluent in this kind of conversations and I'm curious :D 
Also, I'm pretty sure I have my z hop type set to Slope in all the settings available for that, I'm not at home to check that.

But if the problem is already being tested for a solution I'm very happy to hear that, thanks!