bambulab/BambuStudio #5648: Finish time isn't displayed

HodeiDz, 2024-12-29:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Additional system information

CPU: 8th Gen Intel i5-8250U
RAM: 12GB
GPU: GeForce MX130

### Printer

Bambulab A1

### How to reproduce

While i was printing an object y entered in device to see the new feature of finish time but  it wasn't displayed. Then I changed the screen scale to 125% (previously was on 100%) and the finish time now is displayed correctly.

### Actual results

So, depend of the scale of the screen the finish time is diplayed or not

### Expected results

Show the finish time even if the scale is on 100%

### Project file & Debug log uploads

[filament clip.zip](https://github.com/user-attachments/files/18269335/filament.clip.zip)
[debug_Sun_Dec_29_10_44_06_15132.log.zip](https://github.com/user-attachments/files/18269336/debug_Sun_Dec_29_10_44_06_15132.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Rich-b1, 2025-01-25:
This is also occurring on Bambu Studio Beta version 1.10.2.64!

Same issue! If Windows scale is set to 100% then Finish time is not displayed

[image]

If Windows scale is set to 125% then Finish Time is displayed ok:

[image]

DanBao-Bambu, 2025-02-03:
I understand this issue now, and I will resolve it as soon as possible.