bambulab/BambuStudio #5651: Chamber Fan Not Automatically Turning on

Allexkramer432, 2024-12-29:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1 C

### How to reproduce

1. Start a print using PLA
2. Bed temp set to 55 degrees
3. Chamber fan stays at 0%

### Actual results

Chamber fan is not automatically turning on. However, manual turning works.

### Expected results

Gcode settings are default for the Bambu PLA, which means the chamber fan should be turning on.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18269829/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Allexkramer432, 2024-12-31:
Additionally, I've found that rolling firmware back two versions allows the fan to turn on automatically. Two latest firmware versions do not. Plus, printing from the internal models on the printer, such as the benchy, causes the automatic turning on of the fan, even on current firmware.

Allexkramer432, 2025-01-01:
Additional information: If I remove this section from the machine_start_gcode, the fan will come on automatically. I cannot find what this field is:
M142 P1 R35 S40
[image]

XunZhangBambu, 2025-01-21:
@Allexkramer432  M142 command will enable the chassis fan to perform automatic cooling, reducing the speed to minimize noise when the chamber temperature is not high. The heated bed temperature has risen to 55, but the chamber temperature may not have reached the threshold for the fan to turn on yet.