bambulab/BambuStudio #5655: Printer is online but didn't connect to bambustudio,

Thorr28, 2024-12-30:
### Bambu Studio Version

01.10.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

bambu lab A1 / 3DP-039-389

### How to reproduce

install bambulab and connect the printer. LAN connection with access code or PIN with internet both works same, bambustudio cannot connect while Handy app works without any issue.

### Actual results

Mon Dec 30 11:51:34:test bambulab start...
Mon Dec 30 11:51:34:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.10.01.50&guide_version=00.00.00.01
Mon Dec 30 11:51:34:test bing start...
Mon Dec 30 11:51:34:[test_bing]: url=http://www.bing.com/
Mon Dec 30 11:51:34:test http start...
Mon Dec 30 11:51:34:test storage start...
Mon Dec 30 11:51:34:[test_oss]: url=http://upload-file.bambulab.com
Mon Dec 30 11:51:34:test storage upgrade start...
Mon Dec 30 11:51:34:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Mon Dec 30 11:51:34:test storage download start...
Mon Dec 30 11:51:34:test plugin download start...
Mon Dec 30 11:51:34:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.10.01.00
Mon Dec 30 11:51:34:test bing ip resolved = 95.100.146.27
Mon Dec 30 11:51:34:test bing ok
Mon Dec 30 11:51:34:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.10.01.01/8f52eddd78/win_01.10.01.01.zip
Mon Dec 30 11:51:34:test http ok
Mon Dec 30 11:51:34:test bambulab ip resolved = 172.64.152.215
Mon Dec 30 11:51:34:test bambulab ok
Mon Dec 30 11:51:34:test storage upgrade failed
Mon Dec 30 11:51:34:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Mon Dec 30 11:51:34:downloading 5%
Mon Dec 30 11:51:34:downloading 10%
Mon Dec 30 11:51:34:downloading 10%
Mon Dec 30 11:51:34:downloading 15%
Mon Dec 30 11:51:34:downloading 20%
Mon Dec 30 11:51:34:downloading 20%
Mon Dec 30 11:51:34:downloading 30%
Mon Dec 30 11:51:34:downloading 25%
Mon Dec 30 11:51:34:downloading 30%
Mon Dec 30 11:51:34:downloading 40%
Mon Dec 30 11:51:34:downloading 35%
Mon Dec 30 11:51:34:downloading 50%
Mon Dec 30 11:51:34:downloading 40%
Mon Dec 30 11:51:34:downloading 45%
Mon Dec 30 11:51:34:downloading 60%
Mon Dec 30 11:51:34:downloading 50%
Mon Dec 30 11:51:34:test storage failed
Mon Dec 30 11:51:34:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Mon Dec 30 11:51:34:downloading 55%
Mon Dec 30 11:51:34:downloading 70%
Mon Dec 30 11:51:34:downloading 60%
Mon Dec 30 11:51:34:downloading 80%
Mon Dec 30 11:51:34:downloading 90%
Mon Dec 30 11:51:34:downloading 100%
Mon Dec 30 11:51:34:downloading 65%
Mon Dec 30 11:51:34:downloading 70%
Mon Dec 30 11:51:34:downloading 75%
Mon Dec 30 11:51:34:test storage download ok
Mon Dec 30 11:51:34:downloading 80%
Mon Dec 30 11:51:34:downloading 85%
Mon Dec 30 11:51:34:downloading 90%
Mon Dec 30 11:51:34:downloading 95%
Mon Dec 30 11:51:34:downloading 100%
Mon Dec 30 11:51:34:test plugin download ok

### Expected results

work as before

### Project file & Debug log uploads

Mon Dec 30 11:51:34:test bambulab start...
Mon Dec 30 11:51:34:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.10.01.50&guide_version=00.00.00.01
Mon Dec 30 11:51:34:test bing start...
Mon Dec 30 11:51:34:[test_bing]: url=http://www.bing.com/
Mon Dec 30 11:51:34:test http start...
Mon Dec 30 11:51:34:test storage start...
Mon Dec 30 11:51:34:[test_oss]: url=http://upload-file.bambulab.com
Mon Dec 30 11:51:34:test storage upgrade start...
Mon Dec 30 11:51:34:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Mon Dec 30 11:51:34:test storage download start...
Mon Dec 30 11:51:34:test plugin download start...
Mon Dec 30 11:51:34:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.10.01.00
Mon Dec 30 11:51:34:test bing ip resolved = 95.100.146.27
Mon Dec 30 11:51:34:test bing ok
Mon Dec 30 11:51:34:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.10.01.01/8f52eddd78/win_01.10.01.01.zip
Mon Dec 30 11:51:34:test http ok
Mon Dec 30 11:51:34:test bambulab ip resolved = 172.64.152.215
Mon Dec 30 11:51:34:test bambulab ok
Mon Dec 30 11:51:34:test storage upgrade failed
Mon Dec 30 11:51:34:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Mon Dec 30 11:51:34:downloading 5%
Mon Dec 30 11:51:34:downloading 10%
Mon Dec 30 11:51:34:downloading 10%
Mon Dec 30 11:51:34:downloading 15%
Mon Dec 30 11:51:34:downloading 20%
Mon Dec 30 11:51:34:downloading 20%
Mon Dec 30 11:51:34:downloading 30%
Mon Dec 30 11:51:34:downloading 25%
Mon Dec 30 11:51:34:downloading 30%
Mon Dec 30 11:51:34:downloading 40%
Mon Dec 30 11:51:34:downloading 35%
Mon Dec 30 11:51:34:downloading 50%
Mon Dec 30 11:51:34:downloading 40%
Mon Dec 30 11:51:34:downloading 45%
Mon Dec 30 11:51:34:downloading 60%
Mon Dec 30 11:51:34:downloading 50%
Mon Dec 30 11:51:34:test storage failed
Mon Dec 30 11:51:34:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Mon Dec 30 11:51:34:downloading 55%
Mon Dec 30 11:51:34:downloading 70%
Mon Dec 30 11:51:34:downloading 60%
Mon Dec 30 11:51:34:downloading 80%
Mon Dec 30 11:51:34:downloading 90%
Mon Dec 30 11:51:34:downloading 100%
Mon Dec 30 11:51:34:downloading 65%
Mon Dec 30 11:51:34:downloading 70%
Mon Dec 30 11:51:34:downloading 75%
Mon Dec 30 11:51:34:test storage download ok
Mon Dec 30 11:51:34:downloading 80%
Mon Dec 30 11:51:34:downloading 85%
Mon Dec 30 11:51:34:downloading 90%
Mon Dec 30 11:51:34:downloading 95%
Mon Dec 30 11:51:34:downloading 100%
Mon Dec 30 11:51:34:test plugin download ok

### Checklist of files to include

- [X] Log file
- [X] Project file

Thorr28, 2025-01-06:
Hello is there any chance that I will get help with this topic?

Thorr28, 2025-01-10:
Hello, how can I contact some support for this network issue?

walterwongbbl, 2025-01-16:
@Thorr28 Hi. Please check your firewall settings and allow Bambu Studio, then try again.
https://wiki.bambulab.com/en/general/printer-network-ports

Thorr28, 2025-01-16:
Hi I tried, even shuted the firewall down for testing, nothing helped, but when I started printer at Saturday it amazingly started to work again, I didn't identify any change in my network settings, looks strange for me...

Double-A-92, 2025-01-21:
I had the same issue, and I found out that it starts connecting properly when I log out of my Account in Bambu Studio. As soon as I login again, it loses connection though.