bambulab/BambuStudio #5657: Connexion's Page on BambuStudio still blank

GuiDjee, 2024-12-30:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

i7-8750H
16Go RAM
GTX-1070

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go to "LOG OUT" from your account
2. and try to reconnect to your account

### Actual results

When I try to Reconnect my account, the connexion's page is blank... even a full re'installation of bambustudio from official website

### Expected results

To be Connect on my account of course :)

### Project file & Debug log uploads

It's not about Files...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuiDjee, 2024-12-30:
[image]

GuiDjee, 2024-12-30:
PROBLEME SOLVE:
My firewall just blocked internet access to msedgewebview2.exe

Haidiye00, 2024-12-31:
Thank you for identifying the issue and providing a solution~

closed 2024-12-31 by Haidiye00