bambulab/BambuStudio #5659: Fails to send print...

spendnspent, 2024-12-31:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1

### Additional system information

Macbook Pro M2 Pro

### Printer

Bambu Lab A1

### How to reproduce

1. Prepare print
2. Preview / slice plate
3. Print plate
4. Send

### Actual results

Print job uploads to cloud server, but progress stops repeatedly at 70%. (Handy app works flawlessly.)

### Expected results

Print job sends to printer.

### Project file & Debug log uploads

[debug_network.zip](https://github.com/user-attachments/files/18278875/debug_network.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

spendnspent, 2024-12-31:
Now, Bambu Studio is crashing for me as well...

ojoojoo, 2025-01-01:
I think this is a problem at Bambulab, when I copy the file manually to the SD card it works without a problem.

[image]

spendnspent, 2025-01-03:
Still having this problem... The Help menu has a network test option, so here is what I am getting...

[image]

ojoojoo, 2025-01-03:
Problem is gone, today everything works as expected, I guess there was a problem at Bambulab before.
I don't think the Windows version has a network test option. But when I test on the printer itself everything is OK.
So do you still have a problem with this?

walterwongbbl, 2025-01-16:
Hi. @ojoojoo 
You can try deleting the old network plugin and then re-download it.

ojoojoo, 2025-01-16:
> Hi. [@ojoojoo](https://github.com/ojoojoo) You can try deleting the old network plugin and then re-download it.

Thanks for your answer, but as stated before the next day the problem was gone, so I can only conclude the problem was at  Bambulab. I changed nothing in the software

spendnspent, 2025-01-16:
The problem was resolved by turning off my Cisco AnyConnect Socket Filter.- Chase