bambulab/BambuStudio #5663: camera view fail on mac

cristianomeloni, 2024-12-31:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.2

### Additional system information

M1 pro max 64Gb Ram

### Printer

Bambulab p1p

### How to reproduce

start printing, go on device, click on play button, than error player is malfunctioning, before the upgrade worked.

### Actual results

nothingm appeare a guide for solve the same problem on windows, no suggestion on mac

### Expected results

view the printer camera

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

c0desurfer, 2025-01-01:
Sama as https://github.com/bambulab/BambuStudio/issues/5642.

bambu123, 2025-01-06:
> Same as #5642<https://github.com/bambulab/BambuStudio/issues/5642>.

Yes, please wait.