bambulab/BambuStudio #5665: Unable to send to P1S over local wifi

KarstenTWolf, 2024-12-31:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.1

### Printer

P1S and X1C

### How to reproduce

1. Click 'Send' after slicing any plate. 
2. Click send in the 'Send to Printer MicroSD card' prompt
3. The progress bar will sit at 10% for a few seconds
4. A popup 'Connect the printer using IP and access code' will popup. Filling this out and the next page which requires the serial number do not resolve and just return you to the original send popup. 

### Actual results

The above failure loop happens on both machines, all firmware and software is the latest available. 

This did not follow and update or change, it effects both a P1S and X1C which are connected and able to control via app and device page in Studio. Re-binding and logging in/out of Studio did not resolve. 

The work around is to take the SD card and manually transfer files like a barbarian. 

### Expected results

I expected this to work.

### Project file & Debug log uploads

[blank file.3mf.zip](https://github.com/user-attachments/files/18283380/blank.file.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Thorr28, 2025-01-02:
Hi did you get any update about this topic? I have the same like from 12.30.24 or so, I got some setting update or what was it and since then I can see my A1 only as online in the devices but Bambu studio doesn't connect to the printer and when I perform network test in help menu the Test storage and Test storage upgrade fail both, all other test's passes... and result is same I have to manualy upload files to SD card, and watch printing in Handy app, but I would like to find solution for Studio...

damian99999, 2025-01-04:
Similar issue here. Unable to send job to printer X1C. BambuStudio application turns locked while sending job to printer. Progressbar sits forever at 70% sometimes it get stuck on 30% sometimes progressbar is not showing at all. Sometimes "Cancel" button works but for 90% not so i have to kill the applicatin in task manager. Because application lock and kill (not closing normally) network access code is not remebered so you have to enter it every time you restar application.
I observed similar behaviour on previous releases, but it was not so frequent. I were able to workaround it.  Hard reset of printer was not helping.  Workaroud was a change any setting in slicer to generate a slightly different job file. On previous version success rate of such workaroud was high.Now i am frustated by trying it again and again and no sucess. Unable to print.
Version 1.10.1.50, Windows.

damian99999, 2025-01-04:
Update. I removed a new one then reinstalled old version 01.09.03.50 which was working for me (more or less... look my workaround description above). A new "Bambu Network plugin" was downloaded from somwhere. I thought: what the h..? And guess what? The same f... issue with old SW and new network plugin! I will try to retreive an old network plugin and give it a try. Stay tuned...

Thorr28, 2025-01-05:
did you find the Bambu Network Plugin somewhere?

damian99999, 2025-01-06:
Network plugin is downloaded by BambuStudio from somwhere. I don't have patience to get to the source.
I found some historic version of software bambu_networking.dll (laptop broken disk) and replaced. No effect. 
However firmware update (btw connecting printer to cloud) helped. After FW update LAN mode is back again. 
I have no clear conclusions.

Thorr28, 2025-01-10:
I'm confident that the problem is not in the printer, but some bambustudio issue, one problem is that bambu studio is not capable upload to cloud, but even when I switch to local LAN, I only see the printer online, but doesn't connect...
this possibly is the root cause for cloud issue, the cloud adress is wrong, but where is located the configuration I cannot find

Fri Jan 10 15:51:14:test storage start...
Fri Jan 10 15:51:14:[test_oss]: url=http://upload-file.bambulab.com
Fri Jan 10 15:51:14:test storage failed
Fri Jan 10 15:51:14:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Fri Jan 10 15:51:17:test storage upgrade start...
Fri Jan 10 15:51:17:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Fri Jan 10 15:51:18:test storage upgrade failed
Fri Jan 10 15:51:18:status=404, body={"error_msg":"404 Route Not Found"}
, error=

walterwongbbl, 2025-01-16:
Hi. @KarstenTWolf  @damian99999  Please take a look at this wiki.
https://wiki.bambulab.com/en/general/printer-network-ports

@Thorr28 
It seems that we cannot access our cloud server. Can you try using a mobile hotspot?