bambulab/BambuStudio #5666: Flow Dynamics Calibration Manual at the bottom is grayed out.

Kochcd, 2024-12-31:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Click calibration tab at the top
select flow dynamics on the left
The manual calibration on the bottom right will not be grayed for a moment but then go gray and not selectable

### Actual results

mannual calibration is grayed and not selectable 

### Expected results

I should be able to click manual calibration

please refer to this reddit post on how others have gotten this to work with this glitch.
https://www.reddit.com/r/BambuLab/comments/1gpqypp/why_is_the_calibration_button_greyed_out_and_cant/

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Semi0ne, 2025-01-18:
Exact same issue here.

zhimin-zeng-bambulab, 2025-01-23:
It has been fixed in v1.10.2 Public Beta.

closed 2025-02-05 by zhimin-zeng-bambulab