bambulab/BambuStudio #5667: Bambu Studio Crashing when sending file with PLA-CF

Broodthought339, 2024-12-31:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 23H2 Build 22631.4602

### Additional system information

Asus Zenbook 14X OLED UX3404 laptop

### Printer

Bambulab P1S

### How to reproduce

- Add a 3D model or add a primitive to a new project.
- Select Bambulab PLA-CF for material choice from AMS. Roll was loaded in Slot 1. Only have Royal Blue color option to test.
- Use default filament settings
- Slice object
- Select "Print Plate" option and hit "Send" when dialog window pops up.

### Actual results

Once "Send" is clicked, Bambu Studio will hang for a moment before crashing. Program will close without any error message or dialog. 
If Bambu Studio is re-opened, an option to restore the previous session will be given. Attempting to send the file after restoring will give the same result.
[image]

### Expected results

If "Send" is clicked, it should then upload the sliced file and send it to the printer. 

### Project file & Debug log uploads

[BambuStudio Crash Log + File.zip](https://github.com/user-attachments/files/18283686/BambuStudio.Crash.Log.%2B.File.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

AkashCherukuri, 2025-02-09:
Seconded, I'm facing a similar issue as well.

It seems to be caused by PLA-CF, because I'm able to slice and send prints using PLA without any issues.
The crash also occurs when Generic PLA-CF is used.

AkashCherukuri, 2025-02-09:
The bug is probaby caused because there is an error in the warbibn that's supposed to pop up when using CF filaments. A workaround that I've found is to change your device's printer parts to use a hardened steel nozzle. I assume you would have switched nozzles IRL because CF is an abrasive filament.

Here's how you can change the printer parts: https://www.reddit.com/r/BambuLab/comments/13xk4lj/comment/l52jucp/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Hope this helps!

walterwongbbl, 2025-02-19:
@AkashCherukuri Thank you for your feedback, we will confirm this issue.

walterwongbbl, 2025-02-19:
@Broodthought339  Is it this file that crashes when sent?

[image]