bambulab/BambuStudio #5669: A fatal error occurred: "Access violation" Please save project and restart the program.

A-Boom25, 2025-01-01:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home

### Additional system information

Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz   3.30 GHz

### Printer

P1S

### How to reproduce

go into Bambu studio and load a model up slice then add filimant changes and slice again. it won't, same goes for multi-colored models they wont slice.  

### Actual results

A fatal error occurred: "Access violation"
Please save project and restart the program.  

wont slice model

### Expected results

model should be sliced. 

### Project file & Debug log uploads

[TETRIS CASE UPDATED PLATES.zip](https://github.com/user-attachments/files/18284146/TETRIS.CASE.UPDATED.PLATES.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

A-Boom25, 2025-01-02:
made some progress, I discovered that when the prime tower is disabled it will slice the model, but when enabled it says access violation. not sure why. and I need to use the priming tower for the prints I'm working on. hope this helps

QingZhangBambu, 2025-01-27:
@A-Boom25 I can not reproduce it. i slice it successfully. Could you upload dump files, it would be helpful.
PS, I use 1.10.2.64. could you try if that version works?

[image]