bambulab/BambuStudio #5672: mac端打开空白

rinoux, 2025-01-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1 (24B91)

### Additional system information

Apple M2 Pro

### Printer

P1S

### How to reproduce

Mac端应用打开空白，打不开模型，新建窗口也是空白，但是工具栏可以显示。

### Actual results

点了更新就出现这样，重新下载安装也不行，怀疑是你们的更新进程阻止了窗口打开。

### Expected results

你说呢？

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18287976/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2025-01-02:
麻烦给一张描述空白状态的截图

rinoux, 2025-01-02:
https://github.com/user-attachments/assets/807e86db-d7ed-4854-9e77-437ef81341eb

bambu123, 2025-01-02:
窗口可能跑到扩展屏幕上了。
删除studio配置文件试试。

rinoux, 2025-01-02:
扩展屏幕上也没有窗口，我试试看

rinoux, 2025-01-02:
删除BambuNetworkEngine.conf好了

yosefl20, 2026-10-11:
我从官网刚下的版本也出现了一样的问题，删除BambuNetworkEngine.conf并没有用