bambulab/BambuStudio #5678: First Layer Inspection Fails to Detect Absence of First Layer

luckygreen, 2025-01-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win11, latest updates

### Printer

X1C

### How to reproduce

1. Print a .step design that you received from somebody on MakerWorld.
2. Import as a .3mfg
3. Print the file with First Layer Inspection enabled.
4. (For reasons unknown and *not* part of this bug report, the printer goes through the entire print process, i.e. the print head moves as if it were printing, but does not extrude any filament other than the calibration lines. I will file the fact that no filament was extruded as a separate bug report. (Printed in transparent TPU).
2. First Layer Inspection is enabled and does inspect the first layer of the print. (No first layer exists on the build plate).
3. First Layer Inspection will give the green light to continue printing, even though there is no first layer at all.

### Actual results

"Printing" continues in the absence of a first layer.

### Expected results

First Layer Inspection should report an error, since no first layer exists.

### Project file & Debug log uploads

[Gasket 1.zip](https://github.com/user-attachments/files/18292239/Gasket.1.zip)
[log.zip](https://github.com/user-attachments/files/18292265/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

shuizhiguo, 2025-01-06:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

luckygreen, 2025-01-06:
Greetings!
Your log upload instructions are likely based on an old version of Bambu Handy. I am using the current version of Bambu Handy for iOS. The screens in your response do not exist. Below are what amounts to the closest in content screens.

While I do have an option to respond to existing ticket numbers, as you can see from the attached screenshot, no support tickets show as associated with my Bambu Account.

Please provide instructions for the current version of Bambu Handy. 

[image]
[image]
[image]

Last time I uploaded printer log files, I had to upload them by pulling the SD card from the printer and uploading manually, because I couldn't figure out another way. If there exists an alternative using Bambu Studio for Win11, please send me that as well as correct instructions for uploading from Bambu Handy.

Thanks!
-- Lucky

luckygreen, 2025-01-06:
Unfortunately, I am unable to upload the ZIPed printer log file, since the size of the ZIP archive exceeds GitHub's supported upload limit. Please let me know how you would like to receive the printer log file.

Meanwhile, here is the, unfortunately useless, video of the event. The video is useless, because the printer's time-lapse feature cut out the entire part of the video where the print head was moving, as if printing the object, but did not extrude filament. I suspect, the timelapse feature automatically removes the time periods where the print head is not extruding filament.

https://github.com/user-attachments/assets/cbf9c066-8172-46eb-a6c6-62639c532e4b

closed 2025-05-07 by BoBoYu-BamBu